knurling-rs/probe-run #356: Troubleshooting missing case where drivers are not installed

vadixidav, 2022-10-04:
I tried the steps in troubleshooting and still got "Probe not found".

The solution turned out to be that this was a fresh computer and I had not installed ST software yet, so I was missing drivers for my ST-Link. I had to install `STSW-LINK009`.

A simple reminder in the troubleshooting for "Probe not found" to check to make sure your drivers are installed would resolve this.

Urhengulas, 2022-10-05:
What operating system are you on?

vadixidav, 2022-10-06:
Windows 11. It is already fixed. I just had to install the drivers for ST-Link. I opened this ticket to track adding driver installation to the readme.

Urhengulas, 2022-10-06:
Yes, I just wanted to add on what operating system this could appear.

commit c667bae7 (2023-01-10): Merge #371

371: Add "missing drivers" to troubleshooting section r=Urhengulas a=Urhengulas

Fixes #356
commit 9eea8218 (2023-01-10): Merge #371

371: Add "missing drivers" to troubleshooting section r=Urhengulas a=Urhengulas

Fixes #356
PR #371 (2023-01-10): Add "missing drivers" to troubleshooting section
Fixes #356
closed 2023-01-10 by bors[bot]