knurling-rs/probe-run #362: Command failed with status SwdApFault

MorganTechngs, 2022-11-09:
**Describe the bug**
Attempting to probe-run and flash a STM32WL55JC1 returns a variety of errors with the probe connection. 

**To Reproduce**
probe-run --chip STM32WL55JCIx --verbose
or cargo-flash --chip STM32WL55JCIx

*Example*
This was run using standard stm32-hal quickstart main and files. Also tried empty main and that gave the same errors.

**config.toml**
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# Change this runner as required for your MCU.
runner = "probe-run --verbose --chip STM32WL55JCIx" # to list chips, run `probe-run --list-chips.`
rustflags = [
#  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
#  "-C", "link-arg=-Tdefmt.x",
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
#  "-C", "link-arg=--nmagic",
]

[build]
# Change this target as required for your MCU.
target = "thumbv7em-none-eabi" # Cortex-M4 and Cortex-M7a (eg F, L4, H7)
#target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+ (eg G)
# target = "thumbv8m.main-none-eabihf" # Cortex-M33F and Cortex-M35F (eg L5, U5)

[alias]
rb = "run --bin"
rrb = "run --release --bin"
rr = "run --release"

Have also tried with and without flip-link, defmt, and nmagic but no difference.

*cargo.toml*
[package]
authors = ["Your name <your@em.ail>"]
name = "project_name"
edition = "2021"
version = "0.1.0"

[dependencies]
defmt = "0.3.0"
defmt-rtt = "0.3.0"
panic-probe = { version = "0.3.0", features = ["print-defmt"] }

cortex-m = "0.7.3"
cortex-m-rt = "0.7.0"

# Change this import as required for your MCU.
stm32-hal2 = { version = "^1.4.5", features = ["wle5", "wlrt"]}
# stm32-hal2 = { version = "^1.3.3", features = ["h743v", "h7rt"]}

# Peripheral access crate
# stm32wl = { version = "0.15.1", optional = true }

# cargo build/run
[profile.dev]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 3 # <-
overflow-checks = true # <-

# cargo test
[profile.test]
codegen-units = 1
[... 95 lines cut ...]
 DEBUG probe_rs::probe::stlink                              > AP ApAddress {
    dp: Default,
    ap: 0x0,
}: MemoryAp(MemoryApInformation { address: ApAddress { dp: Default, ap: 0 }, only_32bit_data_size: false, debug_base_address: 3759140864, supports_hnonsec: false, has_large_address_extension: false, has_large_data_extension: false })
 DEBUG probe_rs::architecture::arm::ap                      > Reading register IDR
 DEBUG probe_rs::architecture::arm::ap                      > Read register    IDR, value=0x84770001
 DEBUG probe_rs::architecture::arm::ap                      > Reading register BASE
 DEBUG probe_rs::architecture::arm::ap                      > Read register    BASE, value=0xf0000003
 DEBUG probe_rs::architecture::arm::ap                      > Reading register BASE2
 DEBUG probe_rs::architecture::arm::ap                      > Read register    BASE2, value=0x0
 DEBUG probe_rs::architecture::arm::ap                      > Reading register CSW
 DEBUG probe_rs::architecture::arm::ap                      > Read register    CSW, value=0x438000d2
 DEBUG probe_rs::architecture::arm::ap                      > Writing register CSW, value=CSW { DbgSwEnable: 1, HNONSEC: 1, PROT: 6, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U8 }
 DEBUG probe_rs::architecture::arm::ap                      > Reading register CSW
 WARN  probe_rs::probe::stlink                              > send_jtag_command 242 failed: SwdApFault
 DEBUG probe_rs::probe::stlink                              > Current device mode: Jtag
thread 'main' panicked at 'This should not be an unwrap...: Probe(ProbeSpecific(CommandFailed(SwdApFault)))', /Users/morgan/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-cli-0.13.0/src/info.rs:74:22
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

This is using STLinkV3 built into the development WL55JC1 board. Programming using STM32CUBE is possible and works fine.
morgan:stm32-nucleo/ (main*) $ probe-rs-cli list                     [16:46:07]
The following devices were found:
[0]: STLink V3 (VID: 0483, PID: 3754, Serial: 0009002E4D46501720383832, StLink)

Any help would be appreciated!

Urhengulas, 2022-12-09:
Thank you for your report. Since the issue also occurs with `cargo-flash` and `probe-rs-cli` I'd assume this is an issue in the `probe-rs` library.

Can you please report the issue there (github.com/probe-rs/probe-rs) and link to this is here?

closed 2023-02-05 by Urhengulas

seanybaggins, 2023-03-02:
In your binary, does your main loop have `asm::wfti()`. I managed to get rid of this problem by commenting out that line. Hope that helps someone.