knurling-rs/probe-run #363: probe-fun doesn't show probe that is shown with dfu-utils

tartavull, 2022-11-25:
I have a stm32H7 board attached to my computer while dfu-utils correctly detects it, probe-run doesn't. Am I doing something wrong?

```
tartavull@nixos ~/c/e/e/stm32h7 (master)> sudo dfu-util --list
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=0200, devnum=10, cfg=1, intf=0, path="3-4", alt=1, name="@Option Bytes   /0x5200201C/01*128 e", serial="200364500000"
Found DFU: [0483:df11] ver=0200, devnum=10, cfg=1, intf=0, path="3-4", alt=0, name="@Internal Flash   /0x08000000/16*128Kg", serial="200364500000"
Found Runtime: [04f2:b729] ver=2739, devnum=2, cfg=1, intf=2, path="3-8", alt=0, name="Camera DFU Device", serial="200901010001"
```
```
tartavull@nixos ~/c/e/e/stm32h7 (master)> sudo probe-run --list-probes
Error: no probe was found.

Common reasons for this are faulty cables or missing permissions.
For detailed instructions, visit: https://github.com/knurling-rs/probe-run/tree/2f138c3#troubleshooting
````

burrbull, 2022-11-25:
It looks like you try to connect directly to microcontroller for debugging without probe. It is impossible.
For debugging you need any probe  (ST-Link, DAPlink, etc.), connected to SWD or JTAG pins.

If you want only flash via DFU (usb), you could use `https://crates.io/crates/cargo-dfu`

Urhengulas, 2022-11-27:
@tartavull: Does @burrbull's comment help you?