knurling-rs/probe-run #364: Probe visible with `probe-run --list-probes` but cannot be used by runner

nxy7, 2022-11-25:
I'm using pico-debug debugger on my Pi Pico W, and my dev setup consists of Win11 + WSL2.
I've flashed debugger onto the board (my understanding is that pico-debug allows using just one pico and debugger runs on one of 2 pico cores), used usb passthrough so USB is usable inside WSL2.
I've also needed to 
`chmod 777 /dev/bus/usb/001/011`
so I can access usb from non-root user.
Now when I run the following commands
`
lsusb
probe-run --list-probes
`
[image]
everything seems fine. Probe is found and I'd expect that I could run my code. However when I try `cargo run --release` I encounter the following.
`Running `probe-run --chip RP2040 target/thumbv6m-none-eabi/debug/cyw43-example-rpi-pico-w`
Error: Probe could not be created

Caused by:
    Probe was not found.`
 I suspect it might have something to do with permissions (maybe for some reason chmod is not enough?) but truth being told i'm not used to linux/embedded enough to solve it myself. Did anyone encounter that error before? Did I make any obvious mistakes?

nxy7, 2022-11-26:
I've also tried running probe-run natively on windows and got the following error
```
Error: Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
error: process didn't exit successfully: `probe-run --chip RP2040 target\thumbv6m-none-eabi\debug\cyw43-example-rpi-pico-w` (exit code: 1)
```

Urhengulas, 2022-11-27:
Nothing obvious I can see at the moment.

Urhengulas, 2022-11-27:
Can you please run `probe-run` with verbose output? Add `-v` or `-vv` to your runner in `.cargo/config.toml`.

nxy7, 2022-12-01:
```
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 20040000, hard_fault: 10019d21 }
└─ probe_run::elf @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\elf.rs:31
(HOST) DEBUG Searching registry for chip with name RP2040
└─ probe_rs::config::registry @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\config\registry.rs:148
(HOST) DEBUG Exact match for chip name: RP2040
└─ probe_rs::config::registry @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\config\registry.rs:159
(HOST) DEBUG Partial match for chip name: RP2040_SELFDEBUG
└─ probe_rs::config::registry @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\config\registry.rs:162
(HOST) DEBUG RAM region: 0x20000000-0x2003FFFF
└─ probe_run::target_info @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\target_info.rs:115
(HOST) DEBUG section `.data` is in RAM at 0x20000000..=0x2000002F
└─ probe_run::target_info @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\target_info.rs:153
(HOST) DEBUG section `.bss` is in RAM at 0x20000030..=0x2000AC5F
└─ probe_run::target_info @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\target_info.rs:153
(HOST) DEBUG section `.uninit` is in RAM at 0x2000AC60..=0x2000B05F
└─ probe_run::target_info @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\target_info.rs:153
(HOST) DEBUG valid SP range: 0x2000B060..=0x2003FFFC
└─ probe_run::target_info @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\target_info.rs:167
(HOST) DEBUG Searching for CMSIS-DAP probes using libusb
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:16
(HOST) TRACE CMSIS-DAP: CMSIS-DAP device with 1 interfaces
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:68
(HOST) TRACE Could not read string for interface 0, skipping
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:88
(HOST) TRACE No HID interface for CMSIS-DAP found.
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:108
(HOST) DEBUG Found 1 CMSIS-DAP probes using libusb, searching HID
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:25
(HOST) TRACE CMSIS-DAP device with USB path: "\\\\?\\HID#VID_1209&PID_2488#8&2fafbe90&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}"
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:128
(HOST) TRACE                 product_string: "CMSIS-DAP"
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:129
(HOST) TRACE                      interface: -1
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:130
(HOST) TRACE Ignoring duplicate CMSIS-DAP (VID: 1209, PID: 2488, Serial: E6614864D3563835, CmsisDap)
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:40
(HOST) DEBUG Found 1 CMSIS-DAP probes total
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:46
(HOST) DEBUG libusb 1.0.26.11724
└─ jaylink @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\jaylink-0.3.0\src\lib.rs:1557
(HOST) DEBUG libusb has capability API: true
└─ jaylink @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\jaylink-0.3.0\src\lib.rs:1566
(HOST) DEBUG libusb has HID access: true
└─ jaylink @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\jaylink-0.3.0\src\lib.rs:1567
(HOST) DEBUG libusb has hotplug support: false
└─ jaylink @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\jaylink-0.3.0\src\lib.rs:1568
(HOST) DEBUG libusb can detach kernel driver: false
└─ jaylink @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\jaylink-0.3.0\src\lib.rs:1569
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-run-0.3.5\src\probe.rs:25
(HOST) TRACE Attempting to open device matching 1209:2488:E6614864D3563835
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:257
(HOST) TRACE Trying device Bus 003 Device 001: ID 05e3:0612
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:273
(HOST) TRACE Error opening: NotSupported
└─ probe_rs::probe::cmsisdap::tools @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\tools.rs:286
(HOST) TRACE Trying device Bus 003 Device 042: ID 1209:2488
[... cut ...]
└─ probe_rs::probe::cmsisdap::commands @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\commands\mod.rs:393
(HOST) TRACE Receive buffer: [01, 00, 07, 00, 00, 04, 02, 00]...
└─ probe_rs::probe::cmsisdap::commands @ C:\Users\nxyt7\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.13.0\src\probe\cmsisdap\commands\mod.rs:393
Error: Failed to get information for AP ApAddress { dp: Multidrop(11002927), ap: 0 }
error: process didn't exit successfully: `probe-run -vv --chip RP2040 target\thumbv6m-none-eabi\debug\cyw43-example-rpi-pico-w` (exit code: 1)
```

yjc801, 2022-12-02:
I'm running into same issue (on Mac), and the error looks the same as what @nxy7 got. Any insight is appreciated!

```
(HOST) DEBUG 0 of batch of 1 items suceeded
└─ probe_rs::probe::cmsisdap @ /Users/yjc/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/mod.rs:204
(HOST) TRACE Transfer status: NACK
└─ probe_rs::probe::cmsisdap @ /Users/yjc/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/mod.rs:215
(HOST) DEBUG DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
└─ probe_rs::probe::cmsisdap @ /Users/yjc/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/mod.rs:605
```

Urhengulas, 2022-12-05:
> Output when running in windows

Can you please also share the output of WSL?

Urhengulas, 2022-12-09:
@nxy7 said:
> I've also tried running probe-run natively on windows and got the following error
>
> ```
> Error: Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
> error: process didn't exit [...]

This issue might be related to https://github.com/probe-rs/probe-rs/issues/1253.

Can you please try to use `probe-run` with `probe-rs` from their `master` branch? To do that, please clone the `probe-run` repository and change both `probe-rs` and `probe-rs-rtt` to their git version:
```diff
// Cargo.toml
[dependencies]
-probe-rs = "0.13"
-probe-rs-rtt = "0.13"
+probe-rs = { git = "https://github.com/probe-rs/probe-rs" }
+probe-rs-rtt = { git = "https://github.com/probe-rs/probe-rs" }
```

overheat, 2022-12-13:
> @nxy7 said:
> 
> > I've also tried running probe-run natively on windows and got the following error
> > ```
> > Error: Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
> > error: process didn't exit successfully: `probe-run --chip RP2040 target\thumbv6m-none-eabi\debug\cyw43-example-rpi-pico-w` (exit code: 1)
> > ```
> 
> This issue might be related to [probe-rs/probe-rs#1253](https://github.com/probe-rs/probe-rs/issues/1253).
> 
> Can you please try to use `probe-run` with `probe-rs` from their `master` branch? To do that, please clone the `probe-run` repository and change both `probe-rs` and `probe-rs-rtt` to their git version:
> 
> ```diff
> // Cargo.toml
> [dependencies]
> -probe-rs = "0.13"
> -probe-rs-rtt = "0.13"
> +probe-rs = { git = "https://github.com/probe-rs/probe-rs" }
> +probe-rs-rtt = { git = "https://github.com/probe-rs/probe-rs" }
> ```

I have the same issue on Ubuntu 22.10 for RP2040 W. And I try this, also `cargo install --path .` to install it.
But same issue:
```
$ cargo run --bin blinky
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `probe-run --chip RP2040 target/thumbv6m-none-eabi/debug/blinky`
Error: Probe could not be created

Caused by:
    0: hidapi error: Permission denied
    1: hidapi error: Permission denied
```

nxy7, 2022-12-14:
Nevermind, i've tried master branch quickly and it doesnt seem to work (same error)

overheat, 2022-12-14:
> After `chmod 777 /dev/hidraw0`, I got
> 
> ```
> Error: Failed to get information for AP ApAddress { dp: Multidrop(11002927), ap: 0 }
> ```

Same issue on MacOS(Apple M1) for Jlink Base v10.1

Urhengulas, 2022-12-14:
Please report that at https://github.com/probe-rs/probe-rs.

jonathanpallant, 2022-12-14:
Natively on Windows you have to change the USB driver for the CMSIS-DAP probe to WinUSB. Please refer to these instructions in our Embedded Workshop: https://embedded-trainings.ferrous-systems.com/installation.html#windows-only-zadig-jlink-driver

I'm afraid I have no experience with USB redirection in WSL on Windows 11 - last time I used WSL was with Windows 10 where there was no USB redirection support at all. If I get time, I will give it a try on my Windows laptop though.

nxy7, 2023-01-21:
Sorry for late response, i got probe-run working on windows (tbh i don't even know how, i've reinstalled windows and tried working with probe-run after some time and it worked) so I'm closing this issue. I don't think there's point in exploring errors on WSL as it doesn't officially support usb passthrough.

closed 2023-01-21 by nxy7

Urhengulas, 2023-01-23:
Thank you for the update @nxy7!