knurling-rs/probe-run #365: Unable to use --chip-description-path option

Dblm0, 2022-11-28:
**Describe the bug**
Using probe-run from revision 8add0cf passing --chip-description-path option throws the error `Deserializing the yaml encountered an error`

**To Reproduce**
I've created https://github.com/Dblm0/probe-run/tree/probe-rs_yaml_fail to reproduce. 

It turns out, something is wrong with probe-rs dependency from `crates.io`. Changing it to be fetched from `git` solves the problem for me. I understand, it's not the `probe-run` issue, but have no idea how this kind of dependency problems have to be solved.

Urhengulas, 2022-12-09:
> **Describe the bug** Using probe-run from revision [8add0cf](https://github.com/knurling-rs/probe-run/commit/8add0cf89438cb619e7ef7a26256805202eabe7e) passing --chip-description-path option throws [...]

@Dblm0 Thank you for your report. It seems that `probe-rs` fixed this issue in their git version, but didn't release a crates.io version yet. We have to wait until they do a new release.

Dblm0, 2023-02-01:
fixed by 37d703c9f4298784ba0933b4bc984bdd5dd2a922