knurling-rs/probe-run #367: Install failed to select a version for the requirement `funty = "~1.2" on Ubuntu 22.10

overheat, 2022-12-09:
**Describe the bug**
failed to select a version for the requirement `funty = "~1.2" on Ubuntu 22.10

**To Reproduce**
$ cargo install --path .

```
$ cargo install --path .
  Installing probe-run v0.3.3 (/home/aaron/probe-run)
    Updating git repository `https://github.com/probe-rs/probe-rs`
...
    Updating crates.io index
error: failed to compile `probe-run v0.3.3 (/home/aaron/probe-run)`, intermediate artifacts can be found at `/home/aaron/probe-run/target`

Caused by:
  failed to select a version for the requirement `funty = "~1.2"`
  candidate versions found which didn't match: 2.0.0, 1.1.0, 1.0.1, ...
  location searched: crates.io index
  required by package `bitvec v0.22.0`
      ... which satisfies dependency `bitvec = "^0.22"` of package `probe-rs v0.12.0`
      ... which satisfies dependency `probe-rs = "^0.12.0"` of package `probe-rs-rtt v0.12.0`
      ... which satisfies dependency `probe-rs-rtt = "^0.12"` of package `probe-run v0.3.3 (/home/aaron/probe-run)`

```

Urhengulas, 2022-12-09:
It seems you are using the git version of `probe-rs`. Do you also have the git version of `probe-rs-rtt`? If not, this could cause the problem.

overheat, 2022-12-12:
Thanks!

`cargo install probe-run` works on Ubuntu 22.10.