knurling-rs/probe-run #368: Erase-all flash option

huntc, 2022-12-11:
Would an option to erase all flash be considered generally useful? The gen 3 nRF52 boards have AP Protect enabled by default, so it might be handy for an optional  erase all to enable subsequent flashing.

Urhengulas, 2022-12-12:
Are you talking about `probe_rs::flashing::erase_all(session: &mut Session) -> Result<(), FlashError>` ([link](https://docs.rs/probe-rs/latest/probe_rs/flashing/fn.erase_all.html)), or something else?

huntc, 2022-12-12:
> Are you talking about `probe_rs::flashing::erase_all(session: &mut Session) -> Result<(), FlashError>` ([link](https://docs.rs/probe-rs/latest/probe_rs/flashing/fn.erase_all.html)), or something else?

I think so. I’m not familiar with the ineternals of probe-rs, but I’m suggesting the equivalent of `nrfjprog —recover`, which also resets UICR.

Urhengulas, 2023-02-05:
@huntc Can you please try it out if it works as expected? Please install and execute `probe-run` like this:

```console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch erase-all
$ probe-run --chip nRF52840_xxAA --erase-all
$ #                ^^^^^^^^^^^^^ adapt this to your chip
```

I've added a call to `probe_rs::flashing::erase_all` just before downloading the binary to flash.

huntc, 2023-02-06:
Thanks @Urhengulas ! I've tested it and get the following:

```
probe-run --chip nRF52840_xxAA --erase-all target/thumbv7em-none-eabihf/release/evse
(HOST) INFO  flashing program (37 pages / 148.00 KiB)
(HOST) INFO  success!
Error: Overflow error: Attempting to add 2 bytes to Register value 0xfffffffe
```

Should we be concerned about that error? I'm using an nRF52840-DK (gen 3).

huntc, 2023-02-06:
Oh, actually if the device becomes protected (as per the previous run, above), then I get the following:

```
probe-run --chip nRF52840_xxAA --erase-all ../nrf-boot/target/thumbv7em-none-eabihf/release/evse-boot
Error: A ARM specific error occured.

Caused by:
    An operation could not be performed because it lacked the permission to do so: erase_all
```

An `--erase-all` shouldn't require that permission ,and I can certainly erase the flash via nrfjprog still.

Urhengulas, 2023-02-07:
It probably is my mistake. When creating the session, I forgot to call the `probe_rs::Permissions::allow_erase_all()`. Can you please reinstall `probe-run` from the branch and try again with both protection turned on and off?

If that does not work there is also [probe_rs::Session::sequence_erase_all](https://docs.rs/probe-rs/latest/probe_rs/struct.Session.html#method.sequence_erase_all) we can try.

huntc, 2023-02-08:
Hey @Urhengulas - all now appears well! I can flash the device with AP, and then subsequently use `--erase-all` to flash again. Nice work!

commit eaa599ac (2023-02-22): Merge #380

380: `--erase-all` to erase all nonvolatile memory r=Urhengulas a=Urhengulas

Fixes #368
PR #380 (2023-02-22): `--erase-all` to erase all nonvolatile memory
Fixes #368
closed 2023-02-22 by bors[bot]