knurling-rs/probe-run #374: erase_all permission problem?

AdamGlass, 2023-01-21:
**Describe the bug**

NRF5340 device with a Segger Mini in front of it.

probe-run --list-probes
the following probes were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 0101, Serial: 000801031826, JLink)

yet 

probe-run --chip nRF5340_xxAA target/thumbv8m.main-none-eabihf/debug/uart`
Error: An operation could not be performed because it lacked the permission to do so: erase_all

This all worked before.

**To Reproduce**
probe-run --chip nRF5340_xxAA target/thumbv8m.main-none-eabihf/debug/uart`

**Expected and observed behavior**

Download new image.  Instead it claims permission problem for erase_all which seems almost more like a AP_PROTECT thing and continues to run something previously downloaded.

**Probe details**
You can get this information from [`probe-rs-cli`](https://crates.io/crates/probe-rs-cli). Your microcontroller must be connected to your PC / laptop when you run the command below.

*Example:*

probe-rs-cli list
The following devices were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 0101, Serial: 000801031826, JLink)

**I get the following output with the segger on and target on *OR* off**
probe-rs-cli info --chip nrf5340 
Probing target via JTAG
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
Error showing ARM chip information: The debug probe encountered an error.
ERROR probe_rs_cli::info: Error: Debug Probe Error    
ERROR probe_rs_cli::info: Error: An error specific to a probe type occurred    
ERROR probe_rs_cli::info: Error: USB error while writing data to device: Operation timed out    
Probing target via SWD
Error identifying target using protocol SWD: An error specific to a probe type occurred

**Operating System:**
ubuntu

*Example*:

Basic question is this some state on the device that i need to reset/delete/unprotect?

Urhengulas, 2023-01-23:
Hi @AdamGlass, can you please open this issue in https://github.com/probe-rs/probe-rs/? We don't use `erase_all` in our codebase.

AdamGlass, 2023-01-23:
Will do.  I'm also going to re-evaluate for some cabling problem as the behavior doesn't follow logically at all.

closed 2023-01-23 by AdamGlass

ninjasource, 2023-02-14:
I don't know if this helps or muddies the water but the `erase-all` branch fixes this issue for me: 

```
cargo install --git https://github.com/knurling-rs/probe-run --branch erase-all
```