knurling-rs/probe-run #377: Enable ftdi feature in probe-rs

IridiumXOR, 2023-01-26:
Please, can you enable the `ftdi` feature for `probe-rs` dependency in order to be able to use also FTDI debuggers?
Thanks

Urhengulas, 2023-01-27:
Hi @IridiumXOR,

Do you know if we would want`ftdi` or `ftdi-vendored` (https://github.com/probe-rs/probe-rs/blob/master/probe-rs/Cargo.toml#L26-L27)? I didn't find any documentation about the difference.

IridiumXOR, 2023-01-27:
I think it's a problem of licence [1](https://github.com/tanriol/libftdi1-sys/pull/2#issuecomment-608845859) and [2](https://github.com/tanriol/libftdi1-sys/pull/2#issuecomment-608905473). If I have understood correctly, we would want `ftdi` (`ftdi-vendored` compile a static library in order to circumvent _possible_ license problems (in the second link the authors seems to say that there are no license problems)

Urhengulas, 2023-02-03:
@IridiumXOR Can you try if it works as you expect? Install `probe-run` with:

```console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch update-probe-rs --features ftdi
```

You need to have `libtfdi1` installed, but I am sure you know more about that than I do.

IridiumXOR, 2023-02-10:
I'm looking into the code of branch ftdi, where is the part that enable the ftdi feature of probe-rs?

Urhengulas, 2023-02-11:
Please see the changes in https://github.com/knurling-rs/probe-run/pull/381. You can install it with the instructions above.

commit 646ae980 (2023-02-23): Merge #381

381: Add `ftdi` feature r=Urhengulas a=Urhengulas

To test it install `probe-run` like this:

```console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch ftdi --features ftdi
```

Fixes #377
PR #381 (2023-02-23): Add `ftdi` feature
To test it install `probe-run` like this:

```console
$ cargo install --git https://github.com/knurling-rs/probe-run --branch ftdi --features ftdi
```

Fixes #377
closed 2023-02-23 by bors[bot]