knurling-rs/probe-run #386: Cannot flash to XMC4500

KRACK-BIT, 2023-02-27:
**Cannot run `probe-run` on an XMC4500 microcontroller; memory ranges are apparently incorrect**
I am currently trying to flash a very basic hello world program as a proof-of-concept for running on my microcontroller; however, this fails to run.

**To Reproduce**
Steps to reproduce the behavior:

*Example*
1. Install dependencies (snippet from my dockerfile):
```dockerfile
RUN apt install -y libusb-1.0-0-dev libudev-dev
RUN apt install -y libudev-dev
RUN sudo apt install -y acl less man-db nano
RUN cargo install cargo-generate cargo-binutils
RUN cargo install cargo-flash probe-rs probe-rs-cli probe-run

RUN rustup target add thumbv7em-none-eabihf
RUN rustup component add llvm-tools-preview
RUN rustup component add rustfmt
```
2. Write `example/hello.rs`
``` rust
//! Prints "Hello, world!" on the host console using semihosting

#![no_main]
#![no_std]

use panic_semihosting as _;

use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;

#[entry]
fn main() -> ! {
    hprintln!("Hello, world!");

    // exit QEMU
    // NOTE do not run this on hardware; it can corrupt OpenOCD state
    // debug::exit(debug::EXIT_SUCCESS);

    loop {}
}

```
4. Run it with `cargo run --example hello`

**Expected and observed behavior**
I was expecting it to successfully execute and print `hello world`

Resulting behavior however was

```console
root@996f111a2f2d:/workspaces/kybe-rs# cargo run --example hello
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
     Running `probe-run --chip XMC4500-F100x1024 target/thumbv7em-none-eabihf/debug/examples/hello`
Error: Error while flashing

Caused by:
    No flash memory contains the entire requested memory range 0x0c000000.. 0xc000400.
```

**config.toml**
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip XMC4500-F100x1024"

rustflags = [
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C",
  "link-arg=--nmagic",

  # LLD (shipped with the Rust toolchain) is used as the default linker
  "-C",
  "link-arg=-Tlink.x",
]

[build]
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
```

**Probe details**
``` console
The following devices were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 0101, Serial: 000551022083, JLink)
```

**Operating System:**
Docker Desktop on Windows (i.e. Linux VM on Windows); I've got USB pass-through set up, am doing everything within Linux, and it was working without issue when flashing with GDB, so should be working indistinguishable to a Linux box.

**ELF file (attachment)**
I am happy to upload this, but rn GitHub isn't letting me upload things which are not a specific file type; what is the recommended way of uploading ELF files here?

Urhengulas, 2023-02-27:
Hi @KRACK-BIT,

Did you do all the steps listed in the README of our `app-template` (https://github.com/knurling-rs/app-template)? Which HAL did you specify (maybe share your `Cargo.toml`)?

Is there a specific reason you use `cortex-m-semihosting` and not `defmt`? I don't know how the semihosting works and if it is maybe causing the issue.

KRACK-BIT, 2023-02-27:
> Hi @KRACK-BIT,
>
> Did you do all the steps listed in the README of our `app-template` (https://github.com/knurling-rs/app-template)? Which HAL did you specify (maybe share your `Cargo.toml`)?
>
> Is [...]

I've gone through and switched my code to comply with the template; however, the code still does not work.

- I am now complying with all steps from the app-template
- My current HAL is `xmc4500`; I was also considering using `xmc4-hal`, but the `cortex-m-rt` version conflicts with this.
- I have switched over to using `defmt`, so I am no longer using semihosting

KRACK-BIT, 2023-02-27:
Context as to what the `xmc4-hal` / `defmt-test` conflict is:

```console
    ... required by package `defmt-test v0.3.0`
    ... which satisfies dependency `defmt-test = "^0.3.0"` of package `kybe-rs v0.1.0 (/workspaces/kybe-rs)`
versions that meet the requirements `^0.7` are: 0.7.3, 0.7.0

the package `cortex-m-rt` links to the native library `cortex-m-rt`, but it conflicts with a previous package which links to `cortex-m-rt` as well:
package `cortex-m-rt v0.6.15`
    ... which satisfies dependency `cortex-m-rt = "^0.6"` of package `kybe-rs v0.1.0 (/workspaces/kybe-rs)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the links ='cortex-m-rt' value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `cortex-m-rt` which could resolve this conflict
```

KRACK-BIT, 2023-02-27:
@Urhengulas (or anyone for that matter) know if there are any examples of XMC devices successfully using this template for `probe-run`? It claims to be officially supported by `probe-run`, but I can't find any articles or issues giving a successful example of it running

Urhengulas, 2023-03-01:
> @Urhengulas do you (or anyone who is reading this) know if there are any examples of XMC devices successfully using this template for `probe-run`? It claims to be officially supported by `probe-run`, but I can't find any articles or issues giving a successful example of it running

I am not aware of any.

Urhengulas, 2023-03-01:
Maybe you can open an issue on one of the HALs and ask if they have experience?