knurling-rs/probe-run #392: Failed to demangle defmt symbol ...: missing field `crate_name` when running prope-run 0.3.7 works with 0.3.6

bplaat, 2023-04-04:
**Describe the bug**
I have a small micro-controller Rust project that uses defmt and prope-run. It worked fine with `prope-run 0.3.6`. But when I want to run the program with `cargo run --release` with `prope-run 0.3.7`, I got this error:
```
Error: failed to demangle defmt symbol `{"package":"blink-rust","tag":"defmt_info","data":"Blink interval: {} ms","disambiguator":"11913491891077987529"}`: missing field `crate_name` at line 1 column 113
```

**To Reproduce**
Steps to reproduce the behavior:

*Example*
1. Write `src/main.rs`
``` rust
#![no_std]
#![no_main]

// ..

use {defmt_rtt as _,  panic_probe as _ };

#[entry]
fn main() -> ! {
    loop {
        defmt::info!("Test");
    }
}

```
2. Run it with `cargo run --release`

**Expected and observed behavior**
With `prope-run 0.3.6` this works fine but with `0.3.7` you get this error message:
```console
$ cargo run --release
    Finished release [optimized + debuginfo] target(s) in 0.24s
     Running `probe-run --chip STM32H743ZITx 'target\thumbv7em-none-eabihf\release\blink-rust'`
Error: failed to demangle defmt symbol `{"package":"blink-rust","tag":"defmt_info","data":"Blink interval: {} ms","disambiguator":"11913491891077987529"}`: missing field `crate_name` at line 1 column 113
```

**config.toml**
The contents of your project's `.cargo/config.toml` file:
```toml
[target.thumbv7em-none-eabihf]
runner = 'probe-run --chip STM32H743VITx'
rustflags = [
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
]

[build]
target = "thumbv7em-none-eabihf"

[env]
DEFMT_LOG = "trace"
```

**Cargo.toml**
The contents of your project's `Cargo.toml` file:
```toml
[package]
name = "blink-rust"
version = "0.1.0"
edition = "2021"
include = ["src/main.rs"]

[dependencies]
cortex-m = { version = "0.7.4", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.1"
stm32h7xx-hal = {version = "0.13.1", features = ["stm32h743v","rt"]}
panic-probe = { version = "0.3", features = ["print-rtt"] }
defmt = "0.3.3"
defmt-rtt = "0.4"

[profile.release]
codegen-units = 1
debug = 1
lto = 'fat'
opt-level = 'z'
overflow-checks = false
```

**Probe details**
```console
$ probe-run --list-probes
the following probes were found:
[0]: STLink V2-1 (VID: 0483, PID: 374b, Serial: 0671FF525149787867184947, StLink)
```

**Operating System:**
Windows, on WSL (Ubuntu) I get the same issue

**Additional context**
Maybe I forgot something stupid or maybe the JSON symbol info really is missing a field :|

Urhengulas, 2023-04-05:
This is because we changed the wire format but missed bumping the wire format version. Updating your `defmt` should fix this.

closed 2023-04-05 by bplaat

Slushee-a, 2023-04-15:
The latest version of defmt on crates.io (`0.3.4`) still suffers from this issue.
A temporary workaround is using the `defmt` repository directly. 

`defmt-rtt 0.4` requires `defmt 0.3.4`, so doing `defmt = { git = ... }` won't work. `patch.crates-io` can be used instead.
```toml
[dependencies]
defmt = "0.3.4"
defmt-rtt = "0.4"

[patch.crates-io]
defmt = { git = "https://github.com/knurling-rs/defmt" }
```

Urhengulas, 2023-05-04:
https://github.com/knurling-rs/defmt/pull/750 adds backwards compatibility for wire format 3. This will get released soon.