knurling-rs/probe-run #393: `probe-run` fails to build due to yanked dependency

KizzyCode, 2023-04-05:
**Describe the bug**
`probe-run` fails to build because `defmt-decoder = "=0.3.5"` has been yanked.

**To Reproduce**
Run `cargo install probe-run --version=0.3.7`

**Expected and observed behavior**
Expected: `probe-run` is installed successfully. 
Observed: `probe-run` cannot be installed. 

*Example*: I expected to see a backtrace but instead the `cargo run` command hanged / stopped responding.
``` console
$ cargo install probe-run --version=0.3.7 
    Updating crates.io index
  Installing probe-run v0.3.7
error: failed to compile `probe-run v0.3.7`, intermediate artifacts can be found at `/var/folders/3k/j_by2bw56jd5w_ss9c8qr3kc0000gn/T/cargo-installn96znm`

Caused by:
  failed to select a version for the requirement `defmt-decoder = "=0.3.5"`
  candidate versions found which didn't match: 0.3.6, 0.3.4, 0.3.3, ...
  location searched: crates.io index
  required by package `probe-run v0.3.7`
```

**Operating System:**
macOS Ventura Version 13.3 (22E252)

sureshjoshi, 2023-04-05:
Context: It looks like that version was yanked about 4-5 hours ago. 

https://github.com/knurling-rs/defmt/pull/748

goldenMetteyya, 2023-04-06:
will this be fixed soon?

KizzyCode, 2023-04-06:
@goldenMetteyya If you're asking because you're unable to install the app, there are two workarounds:

1. You can force the installation using the `--locked`-switch (i.e. `cargo install probe-run --locked`) – if this is a good idea or not depends on the reason why `defmt-decoder` has been yanked (haven't looked into this). 

2. You can install an older version (e.g. 0.3.6) using `cargo install probe-run --version=0.3.6`

overheat, 2023-04-11:
```
➜ cargo install probe-run
    Updating crates.io index
  Installing probe-run v0.3.7
error: failed to compile `probe-run v0.3.7`, intermediate artifacts can be found at `/var/folders/7n/r69j66410wvffm42j24q6wnr0000gn/T/cargo-installfdkxiL`

Caused by:
  failed to select a version for the requirement `defmt-decoder = "=0.3.5"`
  candidate versions found which didn't match: 0.3.6, 0.3.4, 0.3.3, ...
  location searched: crates.io index
  required by package `probe-run v0.3.7`
```

Same reason?

closed 2023-04-12 by bors[bot]

justahero, 2023-04-12:
Thanks for raising the issue & all others for providing usable workarounds.

A new version of `probe-run` (version `0.3.8`) has been released now and should fix the dependency problem.

commit 50adeec0 (2023-04-12): Merge #395

395: chore: Bump defmt-decoder r=justahero a=elpiel

Reason: 0.3.5 has been yanked and it's not currently possible to install the latest version of `probe-run`

Fixes #393
PR #395 (2023-04-12): chore: Bump defmt-decoder
Reason: 0.3.5 has been yanked and it's not currently possible to install the latest version of `probe-run`

Fixes #393