knurling-rs/probe-run #419: Cannot install probe-run on Mac M1

adrien3d, 2023-08-12:
**Describe the bug**
While trying to `cargo install probe-run` on my Mac OS 13.5 M1, with `rustc 1.71.1 (eb26296b5 2023-08-03)` I have the following error:

```
   Compiling probe-run v0.3.10
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin:/opt/homebrew/sbin:/Users/adrien/.pyenv/shims:/Users/adrien/.pyenv/bin:/Users/adrien/.nvm/versions/node/v14.17.4/bin:/usr/local/sbin:/opt/homebrew/bin:/opt/homebrew/Cellar:/opt/homebrew/opt/make/libexec/gnubin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/gcc-arm-none-eabi-8-2019-q3-update/bin:/usr/local/nRF-Command-Line-Tools_10_13_0/mergehex:/usr/local/nRF-Command-Line-Tools_10_13_0/nrfjprog:/Users/adrien/pebble-dev/pebble-sdk-4.5-mac/bin:/usr/local/go/bin:/usr/local/mysql/bin:/opt/local/bin:/opt/local/sbin:/Users/adrien/Library/Android/sdk/tools:/Users/adrien/Library/Android/sdk/platform-tools:/Applications/Android Studio.app/Contents/gradle/gradle-4.1/bin/gradle:/Users/adrien/Dev/esp/xtensa-esp32-elf/bin:/Users/adrien/.cargo/bin:/usr/local/flutter/bin:/Users/adrien/.ebcli-virtual-env/executables:/Users/adrien/Dev/go/bin" VSLANG="1033" ZERO_AR_DATE="1" "cc" "-arch" "x86_64" "-m64" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/rustcbMiwXd/symbols.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.0.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.1.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.10.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.11.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.12.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.13.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.14.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.15.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.2.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.3.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.4.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.5.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.6.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.7.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.8.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.probe_run.1318601c985768d9-cgu.9.rcgu.o" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/probe_run-df910ff95ee6aecd.59t23fziqry1vbdn.rcgu.o" "-L" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps" "-L" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/build/hidapi-4dbc982d828065eb/out" "-L" "/opt/homebrew/Cellar/libusb/1.0.26/lib" "-L" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libgit_version-3cf9727ee8f56e4d.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libdefmt_decoder-fac528c896d83d0b.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libserde_json-a4ab7e805f503fb9.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libdissimilar-e633fcadd7374431.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libdefmt_json_schema-a8e16a2a11fe2ea2.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libnom-29eff71e01fbf5aa.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libtime-e4ca3148ec4499d6.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libtime_core-1244924257a08e60.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libderanged-48b481392b4b46ec.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libdefmt_parser-bbba1c0374361232.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libclap-911e91fad70c2608.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libclap_builder-af989f24fbae84d1.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libstrsim-b9eb6c507c5e8dae.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libanstream-657f2ce3df7abfb1.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libanstyle_query-1a6518a8293c226d.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libcolorchoice-982085d03ea25c50.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libanstyle_parse-010120ccc940cfe7.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libutf8parse-f6c59646bf9bb265.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libclap_lex-b9cf28807f348292.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libanstyle-ea0c34e168ac2a4f.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libcolored-03f860d387ded58f.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/liblazy_static-892c64fc7461bb89.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libsignal_hook-fa81b45fabff9b17.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libsignal_hook_registry-9cd4c202dd45f212.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libprobe_rs-3bda3c91701dcae5.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libkmp-8cd043544697d857.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libespflash-01e3fe7330e8e314.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libtoml-575611ba7a3f2dd3.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libtoml_edit-9586b954fb805e91.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libserde_spanned-d68d832cc4efc136.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libwinnow-ea909ba51ba3edb5.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libtoml_datetime-2ba4e0d92ef8d901.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4/release/deps/libsha2-0fb0f9a1743798cb.rlib" "/private/var/folders/ww/lc242p6n5b96bbzp [... cut ...]
                core::ops::function::impls::_$LT$impl$u20$core..ops..function..FnMut$LT$A$GT$$u20$for$u20$$RF$mut$u20$F$GT$::call_mut::h2b6decc5fc2dfbac in libprobe_rs-3bda3c91701dcae5.rlib(probe_rs-3bda3c91701dcae5.probe_rs.682adb8aeacbe238-cgu.5.rcgu.o)
                ...
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `probe-run` (bin "probe-run") due to previous error
error: failed to compile `probe-run v0.3.10`, intermediate artifacts can be found at `/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-install8WJug4`
```

Urhengulas, 2023-08-14:
Hi @adrien3d,

The issue seems `libusb` related. Our dependency `probe-rs` uses `rusb` (which links to `libusb`).

Can you please try if you can install the `probe-rs` cli or if it yields the same error?

```console
$ cargo install probe-rs --features cli
```

adrien3d, 2023-08-15:
I think it yields to a similar error:
`  = note: ld: warning: ignoring file /opt/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
          Undefined symbols for architecture x86_64:
            "_libusb_bulk_transfer", referenced from:
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::send_buffer::ha301833d69cedef2 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.7.rcgu.o)
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::receive_buffer::h25cf13f834bf10a8 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.7.rcgu.o)
                _$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$u20$as$u20$probe_rs..probe..stlink..usb_interface..StLinkUsb$GT$::write::h03966503f02c0835 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.7.rcgu.o)
                _$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$u20$as$u20$probe_rs..probe..stlink..usb_interface..StLinkUsb$GT$::read_swo::h35f43a3f9cf63fe4 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.7.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::drain::h6a7be870b864b134 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.14.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::find_packet_size::h7b81e40d41f187c8 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.14.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::read_swo_stream::hc9b2834bee442ec4 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.14.rcgu.o)
                ...
            "_libusb_claim_interface", referenced from:
                rusb::device_handle::DeviceHandle$LT$T$GT$::claim_interface::h43083a6e41f360f1 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::h28b0bbaf7de07b7f in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.2.rcgu.o
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::haf3846594d3ae947 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.2.rcgu.o
                probe_rs::probe::cmsisdap::tools::open_v2_device::h9bd8d8b85c7c3d22 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.6.rcgu.o)
                rusb::device_handle::DeviceHandle$LT$T$GT$::claim_interface::h9e2c345d24f18baf in libjaylink-3999bb48f4f7fbb1.rlib(jaylink-3999bb48f4f7fbb1.jaylink.1b3c1867b86ffb9-cgu.5.rcgu.o)
            "_libusb_close", referenced from:
                core::ptr::drop_in_place$LT$probe_rs..probe..jlink..JLink$GT$::h03543b7a897e2194 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                core::ptr::drop_in_place$LT$rusb..device_handle..DeviceHandle$LT$rusb..context..Context$GT$$GT$::h1073f86caebd7f49 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h2b02cf2aa320c107 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::hee74007bc748ba92 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::hb1cb231b91708d6c in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.3.rcgu.o)
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::hfdbabfdb967c15a9 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.3.rcgu.o)
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h138c3de003cc7d43 in libjaylink-3999bb48f4f7fbb1.rlib(jaylink-3999bb48f4f7fbb1.jaylink.1b3c1867b86ffb9-cgu.5.rcgu.o)
                ...
            "_libusb_control_transfer", referenced from:
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_control::hde768870025d37af in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_languages::h4f459a70a8bae41f in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_serial_number_string::h5e7509d66f8e8572 in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_languages::ha67f924ddb39bab6 in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.3.rcgu.o)
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_string_descriptor::ha49e128f0c79ad94 (.llvm.9479203647793347770) in libprobe_rs-d7c8a3481b7c36d1.rlib(probe_rs-d7c8a3481b7c36d1.probe_rs.a4907ccb1ec8ab27-cgu.3.rcgu.o)
            "_libusb_exit", referenced from:
[... cut ...]
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::h28b0bbaf7de07b7f in probe_rs-e2cb46439619edba.probe_rs.8ee92f0540fa598a-cgu.2.rcgu.o
                ...
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `probe-rs` (bin "probe-rs") due to previous error
error: failed to compile `probe-rs v0.20.0`, intermediate artifacts can be found at `/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-installhAAQnP``

Urhengulas, 2023-08-16:
Okay, so I think it is a problem with `rusb` or `probe-rs`. Can you please open an issue there?

adrien3d, 2023-08-17:
Thanks, reported [here](https://github.com/probe-rs/probe-rs/issues/1723)

closed 2023-08-22 by adrien3d

Danny-Duck, 2023-12-01:
For those coming here with the same issue

I was missing [libusb](https://github.com/libusb/libusb) 

`brew install libusb` fixed it