knurling-rs/probe-run #420: Running app with RTT on STM32F4 crashes core with unrecoverable exception

ThadHouse, 2023-08-17:
**Describe the bug**
When attempting to use probe-run to deploy an app containing RTT to an STM32F4, the deploy causes the core to lock up.

**To Reproduce**
Clone https://github.com/ThadHouse/Stm32F4RustCrash
cargo run to an STM32F4

**Expected and observed behavior**
Expect process to be started with "Hello, world!" printed from RTT. However, command eventually finishes with 

```
Error: An ARM specific error occurred.

Caused by:
    Timeout occurred during operation.
```

Looking up in the log, when attempting to start the core, the following error is spit out for 5 seconds, which looks to be the timeout for setting up the core.

```
(HOST) ERROR The core is in locked up status as a result of an unrecoverable exception
ΓööΓöÇ probe_rs::architecture::arm::core::armv7m @ C:\Users\thadh\.cargo\registry\src\index.crates.io-6f17d22bba15001f\probe-rs-0.20.0\src\architecture\arm\core\armv7m.rs:674
```

The whole log is in the repo at err.txt.

The error does not reproduce if there is no RTT in the code.

**Probe details**
[0]: J-Link (J-Link Ultra) (VID: 1366, PID: 1020, Serial: 000505103250, JLink)

**Operating System:**
Seen on both Windows and macOS.

**ELF file (attachment)**
[stmapp.txt](https://github.com/knurling-rs/probe-run/files/12365436/stmapp.txt)

(Replace .txt with .elf)

ThadHouse, 2023-08-17:
Doing some debugging, it seems like its something when flashing causing this. If I do the run listed above, 

If I dump the result of `analyze_vector_table`, I get `Stack: 0xe00abe00 Reset: 0x62d780d`, which are completely wrong. If I then run again with `--no-flash`, I get `Stack: 0x2001fbc0 Reset: 0x80001a9` which is correct. Something happening during flash is completely bringing down the debug interface in a way reset_and_halt can't get out of.

burrbull, 2023-08-17:
Could you try flash with another probe? Not J-Link

ThadHouse, 2023-08-17:
I updated the post because its not actually related to RTT. The program without RTT does the same thing, however because of how probe-run is written that never actually results in an error, as its sitting there waiting forever for a halt.

Urhengulas, 2023-08-17:
@ThadHouse Can you please try what happens if you replace `probe-run` with `probe-rs run`? The flags are almost the same.

You can install the `probe-rs` cli with:
```console
$ cargo install probe-rs --features cli
```

And then change your `.cargo/config.toml` like this:
```diff
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
-runner = "probe-run --chip nRF52840_xxAA"
+runner = "probe-rs run --chip nRF52840_xxAA"
```

ThadHouse, 2023-08-17:
probe-rs run results in the behavior as if it’s running without rtt. The core doesn’t actually start, and the verbose logs show the locked up error message, but the deploy app keeps running forever because the wait for halt function with no timeout doesn’t error out if the core status is locked up.

ThadHouse, 2023-08-17:
So some more debugging info, which I got by running a modified probe-run.

Before flash(), I can reset_and_halt the core as often as I'd like. The SP and PC registers get set to the expected values. However after flash() reset_and_halt results in the locked up behavior, with SP and PC being wildly incorrect. Whats interesting is when SP and PC are wrong, they're the same values every time I try to run probe-run. They don't change with each iteration. If I grab SP and PC after flash() but before core.reset_and_halt(), SP and PC point to the root of RAM, which makes sense because the flasher app is still ready to run at that point.

Is there an easier way to find the flasher app that gets deployed, other then to just manually disassemble the yaml. I'm surprised the raw assembly isn't stored somewhere.

I'm trying to dig through errata and see if theres something for the F405 that results in a bad reset when running code from ram. I'd be surprised thouh.

ThadHouse, 2023-08-17:
Turns out this is our fault. Whoever designed the boards left Boot0 floating, and for some reason programming the flash does something that sets Boot0 high, which causes the reset to jump to ram. First 2 words in ram are 0xe00abe00 and 0x62d780d, which are SP and PC after boot.

closed 2023-08-17 by ThadHouse