knurling-rs/probe-run #424: Issues using `probe-run` with an fe310g002 microcontroller

romancardenas, 2023-10-02:
**Describe the bug**
I am not able to flash my binary using `probe-run` over a fe310g002 RISC-V target. However, GDB seems to work fine.

**To Reproduce**

1. Run a binary with `cargo run`

**Expected and observed behavior**
I expected my binary to be successfully flashed and run on my board. However, I get this output:

```
$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.42s
     Running `probe-run --chip fe310-g002 /Users/rcardenas/riscv-rs/riscv-slic/target/riscv32imc-unknown-none-elf/debug/hifive1-test`
Error: Error while flashing

Caused by:
    0: Something during the interaction with the core went wrong
    1: A RISCV specific error occurred.
    2: The target does not support halt after reset.
```

**config.toml**

``` toml
[target.'cfg(all(target_arch = "riscv32", target_os = "none"))']
# runner = "../../../toolchains/sifive/bin/riscv64-unknown-elf-gdb -q -x gdb_init"
runner = "probe-run --chip fe310-g002"
rustflags = [
  "-C", "link-arg=-Thifive1-link.x",
]

[build]
target = "riscv32imc-unknown-none-elf"
```

**Probe details**

``` console
$ probe-run --list-probes
the following probes were found:
[0]: J-Link (HiFive) (VID: 1366, PID: 1061, Serial: 000979006714, JLink)
```

**Operating System:**
MacOS

**ELF file (attachment)**

It does not apply to my bug. It seems to be an issue with the board's debugger.

Urhengulas, 2023-10-03:
Can you please test it with `probe-rs run`? https://probe.rs/docs/getting-started/installation/

```diff
# .cargo/config.toml
-runner = "probe-run --chip fe310-g002"
+runner = "probe-rs run --chip fe310-g002"
```

romancardenas, 2023-10-04:
Sure! I got the following output:
```zsh
cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `probe-rs run --chip fe310-g002 /Users/rcardenas/riscv-rs/riscv-slic/target/riscv32imc-unknown-none-elf/debug/hifive1-test`
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, ha
     Erasing sectors ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )Error: The flashing procedure failed for '/Users/rcardenas/riscv-rs/riscv-slic/target/riscv32imc-unknown-none-elf/debug/hifive1-test'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: A RISCV specific error occurred.
    2: The target does not support halt after reset.
```

Urhengulas, 2023-10-04:
Since the issue happens there as well, it seems to be a problem with the `probe-rs` library. Please raise an issue there: https://github.com/probe-rs/probe-rs

romancardenas, 2023-10-04:
Done: https://github.com/probe-rs/probe-rs/issues/1793

Should I close this issue?

Urhengulas, 2023-10-04:
I will. Thank you

closed 2023-10-04 by Urhengulas