knurling-rs/probe-run #428: Wrong SRAM regions for STM32L4

VilNeo, 2023-10-29:
Hello,

many (every?) STM32L4 devices provide two SRAM regions (SRAM1 and SRAM2) which can be accessed from 0x20000000 continuously. This makes it possible to use e.g. 64k for STM32L432KCU or 40k for STM32L412KBU.

Unfortunately, with the [commit](https://github.com/knurling-rs/probe-rs/commit/e835897d9a1c8dc041251cad77a514b1fbe41d2d) the available SRAM regions switched from (SRAM1+SRAM2) to just SRAM1.

This may lead to some problems:

1. I was not aware of this change (still using e.g. 64k RAM for STM32L432KCU in memory.x) and my program has been behaving [strangely](https://github.com/knurling-rs/defmt/issues/791). Just the [INFO message](https://github.com/knurling-rs/probe-run/blob/00c446531c912d31fde9b830667cbc36eb487d7c/src/main.rs#L142) gave me the right clue. So I would prefer to get a more descriptive error message maybe even with program exit here.
2. Is there an approach for dealing with multiple RAM-sections with probe-rs? If not, I would prefer to have the full SRAM1+SRAM2 available.

Of course, I could change them as well resulting in a pull request. But I would like to have the previously mentioned points discussed beforehand.

Thank you very much and best regards
Alex

BriocheBerlin, 2023-10-30:
Hi there, thanks for the description. However as stated in the README, `probe-run` is in maintenance mode now, so we won't add any behaviour here. We recommend switching to [`probe-rs`](https://probe.rs/), see the migration guide: https://ferrous-systems.com/blog/probe-run-deprecation/

BriocheBerlin, 2023-10-30:
> Unfortunately, with the [commit](https://github.com/knurling-rs/probe-rs/commit/e835897d9a1c8dc041251cad77a514b1fbe41d2d) the available SRAM regions switched from (SRAM1+SRAM2) to just SRAM1.

The linked to commit is in the probe-rs repo, so maybe it's an idea to ask there.

VilNeo, 2023-10-31:
Oh, thank you very much. I was not aware of this.

Best regards
Alex

closed 2023-10-31 by VilNeo