knurling-rs/probe-run #429: Using app-template - Timeout occurred during operation

jarrod817, 2023-11-29:
**Describe the bug**
I am using a raspberry pi pico and started with the tutorial here: https://github.com/knurling-rs/probe-run#troubleshooting
cargo.toml is still like the app-template other than updating the panic-probe per the tutorial:
```panic-probe = { version = "0.2", features = ["print-rtt"] }```

I used this memory.x file:
```
MEMORY {
    BOOT2 : ORIGIN = 0x10000000, LENGTH = 0x100
    FLASH : ORIGIN = 0x10000100, LENGTH = 2048K - 0x100
    RAM   : ORIGIN = 0x20000000, LENGTH = 256K
}
```

**To Reproduce**
Steps to reproduce the behavior:

❯ cargo run --bin hello
    Finished dev [optimized + debuginfo] target(s) in 0.25s
     Running `probe-run --chip RP2040 target\thumbv6m-none-eabi\debug\hello`
(HOST) INFO  flashing program (3 pages / 12.00 KiB)
(HOST) INFO  success!
Error: An ARM specific error occurred.

Caused by:
    Timeout occurred during operation.
error: process didn't exit successfully: `probe-run --chip RP2040 target\thumbv6m-none-eabi\debug\hello` (exit code: 1)

**Expected and observed behavior**
From the documentation I expected to see this:

```
$ cargo run --bin hello
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello`
  (HOST) INFO  flashing program (30.22 KiB)
  (HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO:hello -- Hello, world!
────────────────────────────────────────────────────────────────────────────────
  (HOST) INFO  exiting because the device halted.
To see the backtrace at the exit point repeat this run with
`probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello --force-backtrace`
```

**config.toml**
```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip RP2040"

rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  "-C", "link-arg=--nmagic",
]

[build]
target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+

[alias]
rb = "run --bin"
rrb = "run --release --bin"
```

**Probe details**
``` 
❯ probe-run --list-probes
the following probes were found:
[0]: Debug Probe (CMSIS-DAP) (VID: 2e8a, PID: 000c, Serial: E6614103E7531D2F, CmsisDap)
```

**Operating System:**
Windows 11

**ELF file (attachment)**

Couldn't figure out how to upload this, Github is saying it's an unsupported type.

jonathanpallant, 2023-11-29:
The RP2040 requires a special boot loader which is not included in our example applications because it's highly RP2040 specific. Without it the ROM loader will pull 256 bytes of flash into RAM, and then immediately Hardfault because the code it loaded was not valid T32 code (it looks like you left this part of the chip blank).

I can recommend asking in the rp-rs Matrix chat room at https://matrix.to/#/#rp-rs:matrix.org, or looking at https://github.com/rp-rs/rp2040-project-template.

closed 2023-11-29 by jonathanpallant

jarrod817, 2023-11-29:
Could you point out where it was left blank? Or what is an example of what this bootloader name looks like?